QuickSpot/walter-arduino #1: Board selection in Platform.IO

GeirFrimann, 2023-04-27:
The ESP32S3 Dev Module is not available in Platform.IO. 
I have tried esp32-s3-devkitc-1, but it doesn't work.
Is there any other board that will work, or any work-around?

daanpape, 2023-04-27:
Hi Geir,

The ESP32-S3 should work, we will test it out and reply ASAP. 

Kind regards,
Daan

GeirFrimann, 2023-04-27:
> Hi Geir,
> 
> The ESP32-S3 should work, we will test it out and reply ASAP.
> 
> Kind regards, Daan
 The code compiles with esp32-s3-devkitc-1, but nothing is displayed in the terminal

tomkeuper, 2023-05-31:
I have used the following configuration to compile and upload to Walter.
```
[env:esp32-s3-devkitc-1]
monitor_speed = 115200
platform = espressif32
board = esp32-s3-devkitc-1
upload_speed = 115200
framework = arduino

board_build.partitions = huge_app.csv

build_flags =
    -D ESP32
    -D ARDUINO_USB_CDC_ON_BOOT=1
```

daanpape, 2023-12-06:
Hi Geir,

We finally had some time to test out PlatformIO and we have created this first settings file:

[walter.json](https://github.com/QuickSpot/walter-arduino/files/13583710/walter.json)

Can you please test if this also works for you?

DeepBlue-Dev, 2024-02-09:
I tested the board definition today, and it works!

daanpape, 2024-02-09:
@DeepBlue-Dev thank you for the confirmation. I will now close this issue.

closed 2024-02-09 by daanpape