QuickSpot/walter-arduino #5: UDP: Could not transmit data

GeirFrimann, 2023-05-14:
While awaiting the implementation of the MQTT protocol, I have played a little with the supplied ModemTest project.
First I created an UDP integration on our Thingsboard platform, and modified the IP address and port in the code.
Running the original code works fine, the telemetry value counter are updated in Thingsboard.

I have tried to add a new telemetry value to the dataBuf variable, adding 2 elements.
But the modem.socketSend returns with RX: +CME ERROR: 50.
Here is the modified loop():

```
void loop() {
  dataBuf[6] = counter >> 8;
  dataBuf[7] = counter & 0xFF;
  uint16_t distance = random(30, 500);
  Serial.printf("Distance: %d\n", distance);
  dataBuf[8] = distance >> 8;
  dataBuf[9] = distance & 0xFF;

  if(modem.socketSend(dataBuf, 10)) {
    Serial.printf("Transmitted counter value %d\n", counter);
    counter += 1;
  } else {
    Serial.println("Could not transmit data");
    delay(1000);
    ESP.restart();
  }

  delay(10000);
}
```

Here is the serial output:

```
Successfully configured the socket
TX: AT+SQNSD=1,1,11560,"18.130.130.7",0,11560,1,0,0
RX: OK
Connected to UDP server 18.130.130.7:11560
Distance: 384
TX: AT+SQNSSENDEXT=1,,0
RX: +CME ERROR: 50
TX: AT+SQNSSENDEXT=1,,0
RX: +CME ERROR: 50
TX: AT+SQNSSENDEXT=1,,0
RX: +CME ERROR: 50
Could not transmit data
```

Any suggestions what could cause this exception?

daanpape, 2023-09-15:
Hi @GeirFrimann in the new version of the library this should be fixed, can you please confirm?

lolrobbe2, 2025-02-24:
closing due to lack of response

closed 2025-02-24 by lolrobbe2