QuickSpot/walter-arduino #7: UDP: Could not connect to UDP server socket

tomkeuper, 2023-06-09:
I noticed that the UDP connection regularly cannot be established, normally causing the ESP to reboot. But because a reboot will cause all data to get lost I tried to remove as many of these reboot as possible. (because I try to measure time difference aswell, reboots will wipe this data). 

The problem lies within the `socketConnect()` function which returns false causing the if statement mentioned below to get triggered.

```c++
if(!socketConnect(SERV_ADDR, SERV_PORT)) {
    Serial.println("Could not connect to UDP server socket");
    delay(1000);
    error_count++;
    if (error_count >= 3) ESP.restart();
    return;
}
```

This function is present in the default positioning sketch aswell, I just modified the `ESP.restart()` part to only restart it after 3 unsuccessful tries.

Because of the time calculations that I was doing I was able to create a graph that lists the time between each message which revealed a pattern. 
[image]
This time difference is calculated on a seperate node-red server that receives messages from Walter.

Without the `ESP.restart()` function walter will not start sending data again unless a different error triggers a restart.

tomkeuper, 2023-06-09:
Just a note, the pattern does not necessarily concern the time it takes to send a message but more the fact that the udp connection frequently cannot be established.

tomkeuper, 2023-06-28:
It seems to happen every 6 messages.

[image]

The number is the amount of seconds between messages

daanpape, 2023-06-28:
@jonesdptechnics I think we have a fix pending for this issue or was it already in the release from yesterday?

tomkeuper, 2023-07-21:
What do you mean? Im not sure what you are looking for

jonesdptechnics, 2023-10-03:
Hi Tom,

In the current version, this problem should have been fixed. The socket close URC from the modem was not being taken in account, now it is.

Please let me know if you still have trouble with UDP disconnects.

@Roonie34778 Can you elaborate or create a new issue? Thank you!

tomkeuper, 2023-10-04:
Will do, if I continue to see the error I will reopen this ticket. Thanks!

closed 2023-10-04 by tomkeuper