QuickSpot/walter-arduino #9: HTTP Send / Post example

hernae, 2023-09-23:
Hello,

I was testing the Http Send function, without success. I wrote my own version with the previous WalterModem.cpp, which is working fine though.
```

 if(modem.httpConfigProfile(HTTP_PROFILE, "testwalter.requestcatcher.com")) {
    Serial.print("Successfully configured the http profile\r\n");
  } else {
    Serial.print("Failed to configure HTTP profile\r\n");
  }
  static char data[] = "test1234";
// working fine;
modem.httpSend2(HTTP_PROFILE, WALTER_MODEM_HTTP_SEND_CMD_POST, "/test", reinterpret_cast<uint8_t*>(data), sizeof(data), ctbuf, sizeof(ctbuf)) 
// not working
modem.httpSend(HTTP_PROFILE,"/test",reinterpret_cast<uint8_t*>(data),sizeof(data),  WALTER_MODEM_HTTP_SEND_CMD_POST,  WALTER_MODEM_HTTP_POST_PARAM_TEXT_PLAIN, ctbuf, sizeof(ctbuf)) 
```

 In the WalterModem.cpp the httpSend function uses this to create the AT command:
```
 WalterModemBuffer *stringsBuffer = _getFreeBuffer();
    stringsBuffer->size += sprintf((char *) stringsBuffer->data,
            "AT+SQNHTTPSND=%d,%d,\"%s\",%d,%d",
            profileId, httpSendCmd, uri, dataSize, httpPostParam,stringsBuffer);

    _runCmd(arr((const char *) stringsBuffer->data), "OK", rsp, cb, args,
            completeHandler, (void *) (_httpContextSet + profileId),
            WALTER_MODEM_CMD_TYPE_TX_WAIT, NULL, 0, stringsBuffer);
 // here I tried to change the parameters to already "...,WALTER_MODEM_CMD_TYPE_TX_WAIT, data, dataSize); `
```

My version works when I change it to:

 ```
//basically httpSend2
   _runCmd(arr("AT+SQNHTTPSND=", _atNum(profileId), ",",
                _atNum(httpQueryCmd), ",", _atStr(uri), ",", _atNum(dataSize)), // the length is wrong somehow but length of 8 breaks everything
            "OK", rsp, cb, args, 
            completeHandler, (void *) (_httpContextSet + profileId), WALTER_MODEM_CMD_TYPE_DATA_TX_WAIT, data, dataSize);

```

I would be happy to get an example of the httpSend function using HTTP POST. Otherwise I could also provide my httpSend function, but I don't use the stringsBuffer there.

jonesdptechnics, 2023-10-03:
Hi,

We have fixed this bug. Thank you for your suggestion and input!

Can you try again and let us know if it works?

The data size should be the number of bytes to be sent in the POST, not including the null byte in case you pass a null terminated C string.

Kind regards,
Jonas

hernae, 2023-10-05:
Thanks a lot! It is working fine now:

  static char data[] = "Test!";
 if(modem.httpSend(HTTP_PROFILE,  "/test", reinterpret_cast<uint8_t*>(data), sizeof(data), WALTER_MODEM_HTTP_SEND_CMD_POST,WALTER_MODEM_HTTP_POST_PARAM_TEXT_PLAIN,ctbuf, sizeof(ctbuf))) {
//
}

closed 2023-10-05 by hernae