QuickSpot/walter-arduino #12: Add the board definition for platformio *somewhere*

DeepBlue-Dev, 2024-02-23:
Hi!

I mainly use platformio for development and got everything working thanks to the board definition from #1. 
To make using platformio easier, it could be a good idea to add it to the documentation or somewhere else (maybe with a manual that explains how to add custom board definitions). 

If you need help writing the manual, feel free to email me! (cs4r@duck.com).

Thank you in advance!

daanpape, 2024-09-02:
@tverheyde is going to upload the platformio definition.

daanpape, 2025-06-24:
Because of the ongoing dispute between PlatformIO and Espressif there is no official support for the latest version of the ESP-IDF in PlatformIO. In our Discord channel there is some community support to get Walter working within the PlatformIO environment but we are not maintaining an official PlatformIO board definition.

closed 2025-06-24 by daanpape