QuickSpot/walter-arduino #21: Fatal Error

felmue, 2025-02-06:
Hello guys

I somehow managed to make my Walter crashing at boot time with a Fatal error message.

I was playing around with some custom deep sleep code and my best guess would be that I tried to upload new firmware to the ESP32S3 just in a bad moment.

Note: the file `nasNv.dat` sounds to me like it's GNSS assistance data, however my custom deep sleep code did not use GNSS code. That said, I was running the WalterFeels sketch before starting with my custom deep sleep code.

Anyways, below is the error log I get from Walter when running the ModemPassthrough sketch.

Please let me know if you think there is a way to recover from that.

Thanks
Felix

```
10:29:51.503 > 
10:30:07.094 > ^EXIT: Fatal error
10:30:07.105 > 
10:30:07.105 > ^EXIT: Thu Jan 01 00:00:14  1970 File: evtLog.cc@198
10:30:07.106 > 
10:30:07.106 > ^EXIT: '- LOG - ERROR file /fs/rat/current/nasNv.dat actual size 57456 does not equal max_size 58280 when writing partial update
10:30:07.114 > 
10:30:07.114 > ^EXIT: '
10:30:07.803 > 
10:30:07.804 > +SYSSTART
10:30:07.832 > 
10:30:07.832 > ^EXIT:  1) 1C0E6568
10:30:07.833 > 
10:30:07.833 > ^EXIT:  2) 1C1BB2E8
10:30:07.833 > 
10:30:07.833 > ^EXIT:  3) 1C1BB5D6
10:30:07.834 > 
10:30:07.834 > ^EXIT:  4) 1C12CEDE
10:30:40.186 > 
10:30:40.186 > ^EXIT: Fatal error
10:30:40.196 > 
10:30:40.197 > ^EXIT: Thu Jan 01 00:00:15  1970 File: evtLog.cc@198
10:30:40.198 > 
10:30:40.198 > ^EXIT: '- LOG - ERROR file /fs/rat/current/nasNv.dat actual size 57456 does not equal max_size 58280 when writing partial update
10:30:40.206 > 
10:30:40.206 > ^EXIT: '
10:30:40.914 > 
10:30:40.915 > ^EXIT:  1) 1C0E6568
10:30:40.915 > 
10:30:40.915 > +SYSSTART
10:30:40.944 > 
10:30:40.944 > ^EXIT:  2) 1C1BB2E8
10:30:40.944 > 
10:30:40.944 > ^EXIT:  3) 1C1BB5D6
10:30:40.945 > 
10:30:40.945 > ^EXIT:  4) 1C12CEDE
10:30:40.946 > 
10:30:40.946 > ^EXIT: 8.2.0.3 [60186] by robot-soft at 2023-11-29 18:15:00
```

daanpape, 2025-02-06:
Hi @felmue, I'm sorry to hear that the firmware on the Sequans modem has crashed. I have opened a ticket with Sequans to see what would be the best way to recover. I will keep you up-to-date.

Best regards,
Daan

felmue, 2025-02-06:
Hello @daanpape 

thank you for the quick response and offer to help. I appreciate that.

After writing the issue I noticed that I could type in some AT commands and the modem would actually respond to them in a brief window after the `+SYSSTART` output. So I tried to squeeze in an `AT+SQNSFACTORYRESET` and that brought Walter back. :-)

Thanks
Felix

daanpape, 2025-02-06:
Thank you Felix,

Sequans has just confirmed that this is a known issue and your solution is the suggested workaround. This will be fixed in the next maintenance release of the modem firmware. I will keep this issue open until that firmware is available.

Best regards,
Daan