QuickSpot/walter-arduino #22: configPSM crash

MiK546, 2025-02-12:
I'm trying to enable the modem power save mode for when the ESP32 is put to deep sleep mode. The documentation [here](https://www.quickspot.io/documentation.html#/faq?id=power-management) states you just have to call `modem.configPSM(WALTER_MODEM_PSM_ENABLE);`. But when I call the method the Walter crashes immediately. Crash log here:

```
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.

Core  1 register dump:
PC      : 0x400556d5  PS      : 0x00060f30  A0      : 0x8202dad9  A1      : 0x3fcb5f70  
A2      : 0x00000000  A3      : 0xfffffffc  A4      : 0x000000ff  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x00000000  A9      : 0x3fcb6250  
A10     : 0x00060f23  A11     : 0x00000000  A12     : 0x00060f20  A13     : 0x00000000  
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x400556d5  LEND    : 0x400556e5  LCOUNT  : 0xffffffff  

Backtrace: 0x400556d2:0x3fcb1e20 0x4201ba66:0x3fcb1e30 0x42016355:0x3fcb2150 0x42001f99:0x3fcb2210 0x42001c4f:0x3fcb2780 0x42003a96:0x3fcb27a0 0x4037be46:0x3fcb27c0
```

I've tried calling the method right after startup and after first connecting to the NB-IoT network, but neither work. Is there something else I'm supposed to do first that isn't on the documentation or have I encountered a bug?

daanpape, 2025-02-18:
@roema thanks for the second report. @vThibo is already working on it and I will ask @lolrobbe2, who is working on some automatic modem crash handling in WalterModem at the moment to take a look at this as well.

lolrobbe2, 2025-02-19:
@roema 

> I have the same issue when i enable psm mode without psmTAU and psmActive.
> 
> When i try to enable psm mode with
> 
> ```
>     const char *psmTAU = "00100001";
>     const char *psmActive = "00000101";
>     
>     if(modem.configPSM(WALTER_MODEM_PSM_ENABLE, psmTAU, psmActive, &response)){
>           ESP_LOGW(TAG, "Failed to configure PSM profile");
>           set_error(ERROR_LPSM);
> 
>     }
>     else{
>          ESP_LOGI(TAG, "setting PSM profile successfully");
>     }
> ```
> 
> ESP doesnt crash, but modem returns error and psm mode cannot enabled.

could it be that it should be 

 ```
     const char *psmTAU = "00100001";
     const char *psmActive = "00000101";
     
     if(!modem.configPSM(WALTER_MODEM_PSM_ENABLE, psmTAU, psmActive, &response)){
           ESP_LOGW(TAG, "Failed to configure PSM profile");
           set_error(ERROR_LPSM);
 
     }
     else{
          ESP_LOGI(TAG, "setting PSM profile successfully");
     }
> ```

and that you forgot a ! in the code?

daanpape, 2025-02-22:
This will be fixed with increased checks in the library together with a more easy PSM config using seconds. Also better documentation is being worked on.

lolrobbe2, 2025-05-05:
Closing as this has been fixed in `v1.2.0`. (feel free to reopen if there are any more issues)

closed 2025-05-05 by lolrobbe2