QuickSpot/walter-arduino #108: GM02SP Firmware UE8.2.1.0 Missing GPS Support - Need GPS-Enabled Firmware

ScottCarullo, 2025-09-30:
### Issue
GM02SP modem firmware UE8.2.1.0 does not include GPS/GNSS support despite hardware capability.

### Hardware Information
- **Module:** Walter with GM02SP modem
- **Firmware:** UE8.2.1.0 (Sequans Communications)
- **IMEI:** 016169006438084
- **Feature Set:** standard (no GPS)

### Test Results
All GPS commands return `+CME ERROR: operation not supported`:
- `AT+SQNGNSS=?` → ERROR
- `AT+SQNGNSSCFG=?` → ERROR
- `AT+SQNGNSS?` → ERROR
- `AT+CGPS=?` → ERROR

### Request
According to the Walter documentation and datasheet, the GM02SP includes GNSS capability with GPS and GLONASS support. However, my firmware build does not include these features.

Could you please provide:
1. GPS-enabled firmware for the GM02SP modem
2. Instructions for performing firmware update
3. Confirmation of which firmware versions include GPS support

### Additional Context
The datasheet states: "Integrated LNA and SAW filter for GNSS reception" and "Assisted and non-assisted GNSS with GPS and Galileo constellations"

Thank you for your help!

felmue, 2025-10-01:
Hello @ScottCarullo 

have a look [here](https://www.quickspot.io/documentation.html#/faq/modem-update) about how to update modem software.

Thanks
Felix

devArnoud, 2025-10-01:
Hi @ScottCarullo,

The AT commands you mentioned don’t match the reference set we use for communicating with the GM02SP on Walter. It looks like you might be using the wrong or outdated documentation.

For example, the GNSS-related commands normally look like this:

* `AT+LPGNSSCFG=?`

You can find the reference manual we use daily here:
[https://quickspot.io/docs/file/gm02s_at_commands.pdf](https://quickspot.io/docs/file/gm02s_at_commands.pdf)